Seeed-Studio/grove.py #2: Illegal instruction

lucavallin, 2019-03-27:
If I import this:

```
from grove.factory import Factory 
```

and run `python3 main.py`, I get:

`Illegal instruction`

Any idea why?

turmary, 2019-03-29:
Hi lucavallin,
I tried you statement on Raspbian version below

> pi@raspi-peter:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.1 (stretch)
Release:        9.1
Codename:       stretch

> pi@raspi-peter:~ $ cat /boot/issue.txt
Raspberry Pi reference 2017-11-29
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, b71bf3179310aadf9aaf2ffcdf094a88e488dce8, stage5

`pi@raspi-peter:~ $ cat main.py
from grove.factory import Factory

pi@raspi-peter:~ $ python main.py
pi@raspi-peter:~ $ python3 main.py
`
Not found your errors.

Please tell me more about your environment...

turmary, 2019-04-08:
Hi lucavallin,
We tried to fix the problem.
That's cause by a ARMv7 instruction building for MRAA & UPM, which only applied for RPi2 or above.
while RPI1 or Zero(BCM2835, armv6) could not run well.
Please update the lastest code by refer chapter Online Install of top README.md.

lucavallin, 2019-04-08:
Hi @turmary, thanks, will try ASAP.

lucavallin, 2019-04-13:
I actually still get the error, also the online install script doesn't work for me. Any further information I can provide?

turmary, 2019-04-15:
Hi lucavallin,
Please do steps:

1. Check whether `Online Install` executed with message 'OK'

2. check your library version of libmraa1 and libupm1, with below commands:
The version listed here is the right version I tested.

pi@raspi-peter:~/git/grove-py$ **dpkg -s libmraa1**
Package: libmraa1
Status: install ok installed
Priority: optional
Section: libs
Installed-Size: 183
Maintainer: Tom Ingleby <tom@ewsting.org>
Architecture: armhf
Source: mraa
Version: 1.9.0-git20190404-pi20190404

pi@raspi-peter:~/git/grove-py$ **dpkg -s libupm1**
Package: libupm1
Status: install ok installed
Priority: optional
Section: libs
Installed-Size: 4238
Maintainer: Flavian Costin Manea <flavian.manea@rinftech.com>
Architecture: armhf
Source: upm
Version: 1.6.0-git20190404-pi20190404

lucavallin, 2019-04-17:
Hi @turmary,
the script ends in error, but here are my libs:

```
pi@raspberrypi:~ $ dpkg -s libmraa1
Package: libmraa1
Status: install ok installed
Priority: optional
Section: libs
Installed-Size: 183
Maintainer: Tom Ingleby <tom@ewsting.org>
Architecture: armhf
Source: mraa
Version: 1.9.0-git20190404-pi20190404
Depends: libc6 (>= 2.15), libjson-c3 (>= 0.10)
Description: userspace I/O library (runtime)
 mraa (libmraa) is library for interacting with userspace I/O on GNU/Linux
 platforms. The library abstracts platform quirks away from the end user.
 Currently interfaces supported include GPIO/PWM/I2C/SPI.
 .
 This package contains the library used at runtime.
Homepage: http://iotdk.intel.com/docs/master/mraa/
```
and
```
pi@raspberrypi:~ $ dpkg -s libupm1
Package: libupm1
Status: install ok installed
Priority: optional
Section: libs
Installed-Size: 3794
Maintainer: Flavian Costin Manea <flavian.manea@rinftech.com>
Architecture: armhf
Source: upm
Version: 1.6.0-git20180813-pi20180813
Depends: libc6 (>= 2.15), libgcc1 (>= 1:3.5), libjpeg62-turbo (>= 1.3.1), libmraa1, libstdc++6 (>= 6)
Description: Sensor/Actuator repository for MRAA (runtime)
 UPM is a high level repository for sensors that use MRAA. Each sensor
 links to MRAA and are not meant to be interlinked although some
 groups of sensors may be.
 .
 This package contains the library used at runtime.
Homepage: http://iotdk.intel.com/docs/master/upm/
```

turmary, 2019-04-18:
Hi @lucavallin,
The libupm1 is old version(armv7 only) cause by script installing error.
You need a success script installing,
Or at least a installing of libupm1.
    `sudo apt-get install libupm1`

lucavallin, 2019-04-23:
@turmary Seems to work now, many thanks!

turmary, 2019-04-24:
Will close this issue, please open a new issue if you have further questions.

closed 2019-04-24 by turmary