Seeed-Studio/grove.py #11: issue with UPM: No module named UPM

chrisovergaauw, 2019-08-29:
Hi,

I have an rpi model 3b
`pi@mark2:~/python/grove.py $ uname -a
Linux mark2 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux`

`pi@mark2:~/python/grove.py $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.8 (stretch)
Release:	9.8
Codename:	stretch`

The install script seems to run succesffully:
```
#######################################################

  Lastest Grove.py from github install complete   !!!!!

#######################################################
```

mraa and upm are installed on my pi:
```
pi@mark2:~/python/grove.py $ dpkg -s libupm1
Package: libupm1
Status: install ok installed
Priority: optional
Section: libs
Installed-Size: 4238
Maintainer: Flavian Costin Manea <flavian.manea@rinftech.com>
Architecture: armhf
Source: upm
Version: 1.6.0-git20190404-pi20190404
Depends: libc6 (>= 2.15), libgcc1 (>= 1:3.5), libjpeg62-turbo (>= 1.3.1), libmraa1, libstdc++6 (>= 6)
Description: Sensor/Actuator repository for MRAA (runtime)
 UPM is a high level repository for sensors that use MRAA. Each sensor
 links to MRAA and are not meant to be interlinked although some
 groups of sensors may be.
 .
 This package contains the library used at runtime.
Homepage: http://iotdk.intel.com/docs/master/upm/
pi@mark2:~/python/grove.py $ dpkg -s libmraa1
Package: libmraa1
Status: install ok installed
Priority: optional
Section: libs
Installed-Size: 183
Maintainer: Tom Ingleby <tom@ewsting.org>
Architecture: armhf
Source: mraa
Version: 1.9.0-git20190404-pi20190404
Depends: libc6 (>= 2.15), libjson-c3 (>= 0.10)
Description: userspace I/O library (runtime)
 mraa (libmraa) is library for interacting with userspace I/O on GNU/Linux
 platforms. The library abstracts platform quirks away from the end user.
 Currently interfaces supported include GPIO/PWM/I2C/SPI.
 .
 This package contains the library used at runtime.
Homepage: http://iotdk.intel.com/docs/master/mraa/
```

Yet when I try to run `grove_temperature_sensor` I get this error:
```
Traceback (most recent call last):
  File "/usr/local/bin/grove_temperature_sensor", line 11, in <module>
    load_entry_point('grove.py==0.6', 'console_scripts', 'grove_temperature_sensor')()
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 487, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2728, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2346, in load
    return self.resolve()
  File "/usr/local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2352, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.7/site-packages/grove/grove_temperature_sensor.py", line 31, in <module>
    from grove.factory import Factory
  File "/usr/local/lib/python3.7/site-packages/grove/factory/__init__.py", line 2, in <module>
    from .factory import *
  File "/usr/local/lib/python3.7/site-packages/grove/factory/factory.py", line 38, in <module>
    from grove.temperature import *
  File "/usr/local/lib/python3.7/site-packages/grove/temperature/__init__.py", line 2, in <module>
    from .mcp9808 import TemperMCP9808
  File "/usr/local/lib/python3.7/site-packages/grove/temperature/mcp9808.py", line 38, in <module>
    from upm.pyupm_mcp9808 import MCP9808
ModuleNotFoundError: No module named 'upm'
```

this is with version 0.6.
Is there any way you can help me with this?

Kind regards,
Chris

Pillar1989, 2019-08-29:
The latest debian Buster system is completely working,Your system looks like python's various dependencies are imperfect

chrisovergaauw, 2019-09-04:
Thank you. purging and reinstalling my python installation fixed the issue.

closed 2019-09-04 by chrisovergaauw