Seeed-Studio/grove.py #12: Problem opening multiple i2c busses at the same time

nikolozka, 2019-09-23:
in i2c.py the line #51

`Bus.instance = smbus.SMBus(bus)`

causes problems when trying to open two distinct i2c busses from the same app. in our case both calls would open a device on the same i2c bus.

changing to 

`self.instance = smbus.SMBus(bus)`

fixed the problem.

using respberry py at the paj7620 sensor on i2c bus 1 & 3

Pillar1989, 2019-10-18:
This problem solved, do you still need help? @nikolozka

nikolozka, 2019-10-26:
Hey,

Appreciate it.
No further help required.

Best,
Nick

closed 2019-10-26 by nikolozka

maujac, 2022-04-27:
The issue is still in the code. I would like to submit a PR for it, but I'm trying to understand the reasoning behind this.
Why is `instance` being used as a class attribute? It effectively makes the `Bus` class a Singleton.
Are there any advantages to doing this that I'm missing? Thanks.