Seeed-Studio/grove.py #14: curl installation fails due to libmraa1 and libbma456

fjp, 2019-11-16:
When I try to use the online option (curl) to install grove.py I get the following error:

```
fjp@ubuntu:~/git/grove.py$ curl -sL https://github.com/Seeed-Studio/grove.py/raw/master/install.sh | sudo bash -s -
Warning: apt-key output should not be parsed (stdout is not a terminal)
Hit:1 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease                                       
Hit:3 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease                                     
Hit:4 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease                
Hit:5 https://seeed-studio.github.io/pi_repo stretch InRelease                      
Hit:6 http://ports.ubuntu.com/ubuntu-ports bionic-proposed InRelease
Hit:7 http://ppa.launchpad.net/oibaf/graphics-drivers/ubuntu bionic InRelease
Hit:8 http://packages.ros.org/ros/ubuntu bionic InRelease     
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://seeed-studio.github.io/pi_repo stretch InRelease' doesn't support architecture 'arm64'
Enable I2C interface ...
I2C interface enabled...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-rpi.gpio is already the newest version (0.6.3-1ubuntu4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-rpi.gpio is already the newest version (0.6.3-1ubuntu4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libmraa1
dpkg-query: package 'libmraa1' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libmraa1
dpkg-query: package 'libmraa1' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libmraa1
dpkg-query: package 'libmraa1' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
-------------------------------------------------------
     Grove.py installation FAILED, FAILED, FAILED      
-------------------------------------------------------
```

I think this happens because `libmraa1` cannot be located. 

Installing libmraa from [here](https://github.com/intel-iot-devkit/mraa#installing-on-ubuntu) with the following commands:

```
sudo add-apt-repository ppa:mraa/mraa
sudo apt-get update
sudo apt-get install libmraa2 libmraa-dev libmraa-java python-mraa python3-mraa node-mraa mraa-tools
```

After that, I executed the curl command again and got a little further but the installation of grove.py stilll fails because of `libbma456`:

```
fjp@ubuntu:~/git/grove.py$ curl -sL https://github.com/Seeed-Studio/grove.py/raw/master/install.sh | sudo bash -s -
[... 108 lines cut ...]
Reading state information... Done
E: Unable to locate package libbma456
dpkg-query: package 'libbma456' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libbma456
dpkg-query: package 'libbma456' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libbma456
dpkg-query: package 'libbma456' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
-------------------------------------------------------
     Grove.py installation FAILED, FAILED, FAILED      
-------------------------------------------------------
```

I couldn't find out how to install the missing dependency `libbma456`. Any ideas how I can fix this? Thank you!

ecobra, 2020-09-24:
I had the same problem then I put _sudo_  before the _curl_ and the problem was solved.

bigbearishappy, 2022-03-08:
> When I try to use the online option (curl) to install grove.py I get the following error:
>
> ```
> fjp@ubuntu:~/git/grove.py$ curl -sL https://github.com/Seeed-Studio/grove.py/raw/master/install.sh | [...]

Hello, this command works fine now. 
Closing the issue~

closed 2022-03-08 by bigbearishappy