Seeed-Studio/grove.py #15: Installation of python3-mraa fail on Debian GNU/Linux 10 (buster) on docker container

darthkurak, 2019-12-07:
Hello. I want to use GrovePi on my Raspberry Pi 4 on top of IoT Edge runtime (docker engine).
I prepared python module, but have problems with installing all dependencies in dockerfile.
I am using: arm32v7/python:3.7-buster as base image. Image using Debian GNU/Linux 10 (buster) as OS.
Using instructions of both auto and manual install of dependencies, I end up with error:

```
The following packages have unmet dependencies:
 python3-mraa : Depends: libpython3.5 (>= 3.5.0~b1) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

The same instructions work well directly on Raspberry Pi 4. (which is Raspbian GNU/Linux 10 (buster)) 

I know that this is due to some python libraries missing - but both distributions have 3.7.5. What is done extra in Raspbian distribution that this instructions works, and on pure buster, not?

How to fix that? 
Thanks in advance for help :)

darthkurak, 2019-12-07:
I tried this also on python:3.5-buster image (3.5.9) but this still occur :(

xAlstrat, 2020-09-01:
I managed to solve it. Just use the normal python installation version, not the python3 version. Debian buster image uses python3 as python, same for pip.

This is my Dockerfile:

```
FROM arm32v7/python:3.7-slim

ENV PYTHONUNBUFFERED 1

WORKDIR /code/
RUN apt-get update && \
    apt-get install -y tcl tk gcc make curl gnupg git && \
    echo "deb https://seeed-studio.github.io/pi_repo/ stretch main" | tee /etc/apt/sources.list.d/seeed.list && \
    curl https://seeed-studio.github.io/pi_repo/public.key | apt-key add - && \
    apt-get update && apt-get install -y python-rpi.gpio python-mraa python-upm && \
    git clone https://github.com/Seeed-Studio/grove.py && pip install ./grove.py
COPY ./requirements.txt /code/requirements.txt
RUN pip install --upgrade pip && pip install -r /code/requirements.txt
COPY . /code/

CMD python /code/main.py
```

I had to install `git` and `curl` (ignore some libs like `tcl` and `tk`, those are specific to my project).

mgfink, 2021-05-13:
I found this thread as we're also trying to run a script that makes use of grove.py from within a container.  However, the above Docker file isn't working and have been trying to hack with different images, python versions, etc. to no avail.  

I'm running a Raspberry Pi 4 with Raspbian GNU/Linux 10 (buster).  When I run bash on the container in privileged mode and try running the script, I get:

root@0512ceae4514:/code# python script.py
Traceback (most recent call last):
  File "script.py", line 5, in <module>
    from grove.factory import Factory
  File "/usr/local/lib/python3.7/site-packages/grove/factory/__init__.py", line 2, in <module>
    from .factory import *
  File "/usr/local/lib/python3.7/site-packages/grove/factory/factory.py", line 38, in <module>
    from grove.temperature import *
  File "/usr/local/lib/python3.7/site-packages/grove/temperature/__init__.py", line 2, in <module>
    from .mcp9808 import TemperMCP9808
  File "/usr/local/lib/python3.7/site-packages/grove/temperature/mcp9808.py", line 38, in <module>
    from upm.pyupm_mcp9808 import MCP9808
ModuleNotFoundError: No module named 'upm'

I confirmed the python-upm package is installed and attempted installing python3-upm and got the same error @darthkurak posted:

root@0512ceae4514:/code# apt install python3-upm
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-upm : Depends: libpython3.5 (>= 3.5.0~b1) but it is not installable
E: Unable to correct problems, you have held broken packages.

Any suggestions would be greatly appreciated!

Thanks!!

is-qian, 2024-09-21:
Refer to https://github.com/Seeed-Studio/grove.py/issues/73 .This issue is closed, If you have any questions, you can comment and reply.

closed 2024-09-21 by is-qian