Seeed-Studio/grove.py #18: Can't install lib on Jetson Nano

shijiong, 2019-12-31:
Hello,

When I use "curl -sL https://github.com/Seeed-Studio/grove.py/raw/master/install.sh | sudo bash -s -" on Jetson nano, the output informs me as:

Err:18 https://seeed-studio.github.io/pi_repo unknown Release                  
404  Not Found [IP: 185.199.109.153 443]

Any suggestion?

Thanks!

andreagavazzi, 2020-01-20:
Same issue for me. I went for the manual setup and it worked.
Unfortunately it seems they stopped the development for our board :-(

bigbearishappy, 2022-03-08:
Hello.
The website [https://seeed-studio.github.io/pi_repo](https://seeed-studio.github.io/pi_repo) can be accessed normally now.
And you can install it manually by following this [document](https://github.com/Seeed-Studio/grove.py/blob/master/doc/INSTALL.md#install-dependencies)

closed 2022-03-08 by bigbearishappy