Seeed-Studio/grove.py #21: Can not run codes other than the root directory of grove.py

creaturerigger, 2020-02-20:
I installed the package as it is in the documentation but I can't call GPIO of grove.gpio like from grove.gpio import GPIO because it throws an error that No module named grove.gpio etc. But when I copy my py file into the root directory of grove.py and run it, everything seems ok and the code works. What is the problem with that? Shouldn't I be able to run my codes which call grove classes/modules from any location on a Pi?

tang295, 2021-02-03:
If you run "pip3 list", can you see grove.py in there?

tyihuang, 2021-04-09:
You need to see your path, and whether you follow the installation process below

Sonn567, 2021-05-13:
Did you install the packages as sudo or as normal user?
So if I install it as normal user pi grove.py will be saved in /home/pi/.local/lib/python3.7/site-packages
And if you install it as sudo it will be saved in /usr/local/lib/python3.7/dist-packages
And if you installed it with sudo and want to run your python script as normal user then it looks for the pip package in /home/pi/.local/python3.7/site-packages but it is stored in /usr/local/lib/python3.7/dist-packages. Run your script as sudo and it will find the grove.gpio etc. Or you can install the packages for the normal user to. :)
I advise you to install the pip packages with **pip3 install grove.py** and not manually with the github clone.
Run **pip list** or **sudo pip list** and you should see grove.py in the list.

Sonn567, 2021-05-13:
> I installed the package as it is in the documentation but I can't call GPIO of grove.gpio like from grove.gpio import GPIO because it throws an error that No module named grove.gpio etc. But when I [...]

Sonn567, 2021-05-13:
Did you install the packages as sudo or as normal user?
So if I install it as normal user pi grove.py will be saved in /home/pi/.local/lib/python3.7/site-packages
And if you install it as sudo it will be saved in /usr/local/lib/python3.7/dist-packages
And if you installed it with sudo and want to run your python script as normal user then it looks for the pip package in /home/pi/.local/python3.7/site-packages but it is stored in /usr/local/lib/python3.7/dist-packages. Run your script as sudo and it will find the grove.gpio etc. in /usr/local/lib/python3.7/dist-packages. Or you can install the packages for the normal user to. :)
I advise you to install the pip packages with pip3 install grove.py and not manually with the github clone.
Run pip list or sudo pip list and you should see grove.py in the list.

creaturerigger, 2021-05-17:
Thank you all, it's been a long time I haven't been active here.  I will try to rerun the code and I will let you know if it works.

WildJames56, 2021-05-21:
I run the pip3 install grove.py and it run the installation but after when I tap pip list or sudu pip list I dont see grove.py I only see grovepi. Do you have any idea how to fix this?

Sonn567, 2021-05-21:
> I run the pip3 install grove.py and it run the installation but after when I tap pip list or sudu pip list I dont see grove.py I only see grovepi. Do you have any idea how to fix this?

Sorry... Error in my comment. If you installed with pip3, run **pip3 list** not pip list. At the raspberry pi pip is for python2 und pip3 for python3.

is-qian, 2024-09-22:
It is recommended to install this library in a virtual environment. Of course, you can also install it in your system, but you'll need to use the --break-system-packages parameter for source installation. You can try again with the latest code, and if you encounter any issues, you can reopen this issue.

closed 2024-09-22 by is-qian