Seeed-Studio/grove.py #23: Issue with one click installation on Ubuntu 18.04 LTS

kenji-iijima191, 2020-05-19:
I want to use Grove Base Hat for Raspberry Pi 3B running Ubuntu 18.04 LTS. 

I tried
curl -sL https://github.com/Seeed-Studio/grove.py/raw/master/install.sh | sudo bash -s -
But it never succeeds with the below error messages,
main line 30: lua command not found
main line 82: dtparam command not found
E: Unable to locate package libmraal
dpkg-query: package 'libmraa1' is not installed and no information is available

I have executed "sudo add-apt universe" so that i see "universe" in /etc/apt/source.line

My goal is to complete installing grove.py so that user application has path to Grove hardware

tang295, 2021-02-03:
Are you using the Ubuntu Core 18 image from the Raspberry Pi Imager utility? If you try the bigger size Server 20 does it work better? It looks like your image is very minimal.
I've got it running on Debian, so it should work.

is-qian, 2024-09-21:
There are currently no plans to support Ubuntu, so we will close this issue for now. If you have any questions, you can comment and reply.

closed 2024-09-21 by is-qian