Seeed-Studio/grove.py #30: Strip Led is not working

kabcasa, 2020-12-18:
using GROVE BASE HAT RPI with Grove - RGB Led ring (16-WS2813 Mini)
On Rpi 3B Buster version
The Led are on PWM slot .
i can't get the leds switch on 
and no error message is given

code :
from grove.i2c import Bus
from rpi_ws281x import PixelStrip
from time import time, sleep
from grove.gpio import GPIO
from RPi import GPIO as io

def light_led(led, pix, r, g, b):
    for pixel in range(pix):
        led.setPixelColorRGB(pixel, r, g, b)

    print("led mode {} {} {}".format(str(r),str(g),str(b)))
    led.show()

led20 = PixelStrip(16, 12) # Grove RGB led connected to grove PWM
led20.begin()

light_led(led20, 16, 55, 0,0)
sleep(5)
light_led(led20, 16, 0, 55, 0)
sleep(5)
led20._cleanup()

tang295, 2021-02-03:
in grove.py there's an example grove_ws2813_rgb_led_strip.py, does this one work?

kabcasa, 2021-02-03:
i test it also it's the same the led doesn't switch on

tyihuang, 2021-04-09:
Can this help you

tyihuang, 2021-04-09:
https://wiki.seeedstudio.com/Grove-LED_ring/

bradnielsen2981, 2021-08-30:
Same, Im trying to work out with the LED Strip. 1. Which Grove Port do you plug into? Why does it mention PIN when it's using a Grove Shield.

is-qian, 2024-10-18:
Sorry for the late reply. Is there any new progress on this issue? Have you tried it on the latest version?

is-qian, 2024-10-25:
I conducted a test in the latest grove.py by inserting the WS2813 Mini into D18 and running the command `sudo grove_ws2813_rgb_led_strip 18`. The LED lit up properly. It also works fine on the PWM PIN . If you still encounter problems despite following the above steps, please describe your operation procedures and reopen this issue.

closed 2024-10-25 by is-qian