Seeed-Studio/grove.py #34: grove water sensor example seems incorrect?

jaydeep987, 2021-03-13:
I tried to follow grove water sensor example shown in `https://github.com/Seeed-Studio/grove.py/blob/master/grove/grove_water_sensor.py`
The below condition is opposite in my case:
```
if sensor.value > 800:
            print("{}, Detected Water.".format(value))
        else:
            print("{}, Dry.".format(value))
```

It prints >800 Detected water when sensor is not in water and prints <800 Dry when sensor is in water..

Is that expected ?

tyihuang, 2021-04-09:
This is because the resistance drops after contact with water, resulting in such data feedback

is-qian, 2024-10-25:
According to my verification, it is indeed the case. This problem has been fixed in this [commit](https://github.com/Seeed-Studio/grove.py/commit/8899b8ebd74200856b6dabe712ee4e3e4ae34513). Thanks for your reminder.

closed 2024-10-25 by is-qian