Seeed-Studio/grove.py #36: grove_imu_9dof_icm20600_ak09918 module seems to not work

baterja, 2021-03-20:
I'm a pretty new user of Grove modules. I have successfully run 16x2 LCD and ultrasonic ranger with that Python library but I can't run IMU9DOFICM20600 (gyro/accelerometer/compass) with it.

What I try:
```python
from grove.grove_imu_9dof_icm20600_ak09918 import GroveIMU9DOFICM20600

imu = GroveIMU9DOFICM20600()
```

and just instantly I get an error message being logged: `Failed to open i2c bus /, error = -1`. It's not an exception, I can still go on and run methods like `imu.get_accel()` or `imu.get_gyro()` but they keep returning all zeros. That module have no documentation in the library so I don't have an idea if I do something wrong. Did anyone use that module and can confirm it works?

Device info:
Raspberry Pi 2B
Raspbian 10 "Buster"
Python 3.7.3
I use Grove Base Hat for Raspberry Pi.
IMU 9DOF v.2.2 is connected to the first I2C port on the hat.
I have the newest version of the library installed from source.
I have other devices connected to the hat (ultrasonic ranger and LCD) but tried to remove them and it didn't help.
I've run `i2cdetect` and it seems like `68` address is detected.
```bash
pi@raspberrypi:~/work $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- 04 -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- 3e -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --      
```

tyihuang, 2021-04-09:
Don't use the ble client for the time being. If there is a problem, the server can still use it

tyihuang, 2021-04-09:
The content of this page may help you
https://wiki.seeedstudio.com/Grove-IMU_9DOF-lcm20600+AK09918/#fuction-table

is-qian, 2024-10-18:
Sorry for the late reply. Is there any new progress on this issue? Have you tried it on the latest version?

baterja, 2024-10-21:
Heh, hello there after 3 years :) I don't remember if I was able to run it finally, and now I no longer continue that project.

is-qian, 2024-11-25:
Fixed in [commit](https://github.com/Seeed-Studio/grove.py/commit/fd5a5cfffd83fc2fbddfa617efebeef492b13b78)

closed 2024-11-25 by is-qian