Seeed-Studio/grove.py #39: Support for grove led bar

glamorous, 2021-04-01:
Is there a possibility to have a code example for programming the grove led bar?

Pillar1989, 2021-04-08:
Please help him 

主题: [Seeed-Studio/grove.py] Support for grove led bar (#39)

tyihuang, 2021-04-09:
You can think of the relevant information and code here
https://wiki.seeedstudio.com/Grove-LED_Bar/

glamorous, 2021-04-09:
> You can think of the relevant information and code here
> https://wiki.seeedstudio.com/Grove-LED_Bar/

No, that only working when you have the GrovePi, not the Grove Base hat.

gmol, 2021-05-05:
Same request. Please add an example for Grove LED Bar.

glamorous, 2021-07-11:
Nobody who can help? 😞

gughpatrik619, 2022-02-16:
Why is nobody responding? @staff

zolgear, 2023-04-09:
Forked and added a module.

https://github.com/zolgear/grove.py/blob/master/grove/grove_ledbar.py

I referred to "mcauser/micropython-my9221" and the datasheet.

is-qian, 2024-10-18:
Currently, there is no update on this aspect. You can try modifying GrovePi to run it. If you get good results, you can submit a pull request to us for testing.

closed 2024-11-04 by Lesords