Seeed-Studio/grove.py #46: Errors when installing on a Raspberry Pi running Bullseye

jimbobbennett, 2021-11-19:
When running this script on a Raspberry Pi running Bullseye, I get the following errors:

```output
pi@tracker:~ $ curl -sL https://github.com/Seeed-Studio/grove.py/raw/master/install.sh | sudo bash -s -
deb https://seeed-studio.github.io/pi_repo/ stretch main
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
100  3822  100  3822    0     0  15991      0 --:--:-- --:--:-- --:--:-- 15991
OK
Get:1 https://seeed-studio.github.io/pi_repo stretch InRelease [3,167 B]
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease                                                                                
Hit:3 http://archive.raspberrypi.org/debian bullseye InRelease                                                      
Get:4 https://seeed-studio.github.io/pi_repo stretch/main armhf Packages [31.0 kB]
Fetched 34.1 kB in 1s (32.2 kB/s) 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python3-pip

E: Package 'python-pip' has no installation candidate
dpkg-query: package 'python-pip' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python3-pip

E: Package 'python-pip' has no installation candidate
dpkg-query: package 'python-pip' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python3-pip

E: Package 'python-pip' has no installation candidate
dpkg-query: package 'python-pip' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-pip is already the newest version (20.3.4-4+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Enable I2C interface ...
I2C interface enabled...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-rpi.gpio is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-rpi.gpio' has no installation candidate
dpkg-query: package 'python-rpi.gpio' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-rpi.gpio is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-rpi.gpio' has no installation candidate
dpkg-query: package 'python-rpi.gpio' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-rpi.gpio is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-rpi.gpio' has no installation candidate
dpkg-query: package 'python-rpi.gpio' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
-------------------------------------------------------
     Grove.py installation FAILED, FAILED, FAILED      
-------------------------------------------------------
```

This looks like the Python2 packages are not compatible with bullseye, which makes sense as Python2 has been deprecated. It would be nice if the script was update to only install Python3 libraries to avoid these errors.

paul8989, 2021-12-02:
Tried with Bullseye on RPi 4 and the current Seed provided script. Same error here too.

```
E: Package 'python-pip' has no installation candidate
E: Package 'python-rpi.gpio' has no installation candidate
```

Have posted on Seed Community too.

https://forum.seeedstudio.com/t/grove-hat-grove-py-installation-failed-failed-failed/261665/6

RobJWarren, 2021-12-17:
I have the exact same problem, running the install on a Raspberry Pi Zero. Running fine on Buster for over a year.
Would the issue be solved in Github or be best addressed to Seeed Studio?

IcingTomato, 2022-01-20:
#47

RobJWarren, 2022-01-24:
Many thanks. This advice worked a treat and the Grove water detectors are now fully-functional on the Raspberry Pi Zero, running Bullseye.

IcingTomato, 2022-01-28:
Glad to be of assistance! ;-)

ishotjr, 2022-03-06:
> From source code (If you are using Raspberry Pi with Raspberrypi OS >= Bullseye, you have to use this command line.)

Building from source did not work for me on Bullseye:

```
fatal: cannot exec 'remote-https': Input/output error
Fatal Python error: init_sys_streams: can't initialize sys standard streams
Python runtime state: core initialized
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 786, in exec_module
  File "<frozen importlib._bootstrap_external>", line 922, in get_code
  File "<frozen importlib._bootstrap_external>", line 980, in get_data
OSError: [Errno 5] Input/output error
```

I started over in Buster and had no problems at all running the `curl` command. :relieved:

lakshanthad, 2022-03-06:
Hi @ishotjr,

Did you try the commands for Python3?
sudo pip3 install .

I just tried to build from source code on Bullseye (with Python3) and it worked for me. Maybe you could try to reflash Bullseye and try again?

Can I know which Raspberry Pi Model you are using?

Thank you.

ishotjr, 2022-03-06:
@lakshanthad unfortunately I wiped the Bullseye install and replaced it with Buster, so I can't try anything, but here's my exact steps:

```
cd ~
git clone https://github.com/Seeed-Studio/grove.py
cd grove.py
sudo pip3 install .

fatal: cannot exec 'remote-https': Input/output error
Fatal Python error: init_sys_streams: can't initialize sys standard streams
Python runtime state: core initialized
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 786, in exec_module
  File "<frozen importlib._bootstrap_external>", line 922, in get_code
  File "<frozen importlib._bootstrap_external>", line 980, in get_data
OSError: [Errno 5] Input/output error
```

:bowtie:

lakshanthad, 2022-03-06:
Hello @ishotjr,

I followed the exact same steps you have mentioned and worked for me.

My environment is as follows:
- Raspberry Pi 4 Model B
- Raspberry Pi OS Bullseye
- Python 3.9.2
- pip 20.3.4

I would suggest you to reinstall Bullseye and try again. That would help you to start fresh with the installation. If that also gives errors, run the below command to output the complete log and share again here:
sudo pip3 install . > log.txt

Thank you

ishotjr, 2022-03-07:
@lakshanthad ok, thank you! Just to clarify, the previous attempt was a fresh installation too. 😊

lakshanthad, 2022-03-08:
Hello @ishotjr,

I understand. Please try on a fresh install again if you have the time. And if that works well, I can close this issue.

Thank you.

chaeusler, 2022-03-23:
Hi all,

When I try to run i.e. `grove_led` I got the message `Check whether I2C enabled and   Grove Base Hat RPi  or  Grove Base Hat RPi Zero  inserted`. I2C is enabled and the Grove Base HAT sits on top of my Raspberrypi 4. 

For the second attempt I followed the advise from above and started from scratch with a refreshed system Then I ran the commands:
```sh
git clone https://github.com/Seeed-Studio/grove.py
cd grove.py
sudo pip3 install .
```

The output of pip3 install doesn't mention any problems:
```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/pi/grove.py
Collecting smbus2
  Downloading https://www.piwheels.org/simple/smbus2/smbus2-0.4.1-py2.py3-none-any.whl (11 kB)
Building wheels for collected packages: grove.py
  Building wheel for grove.py (setup.py) ... done
  Created wheel for grove.py: filename=grove.py-0.6-py3-none-any.whl size=143445 sha256=04d93e15b483cebfe917875b0c13697f378c31626b82ecf8f0dfe3cc55a40a60
  Stored in directory: /tmp/pip-ephem-wheel-cache-9z15rh2h/wheels/4d/18/97/e09b15ec52bed2c3d11734c763ba83604391bf7651631a48e3
Successfully built grove.py
Installing collected packages: smbus2, grove.py
Successfully installed grove.py-0.6 smbus2-0.4.1
```

Thanks in advance.

lakshanthad, 2022-03-25:
Hello @chaeusler,

I have reproduced your issue on my Raspberry Pi 4 and it works well.

```sh
pi@raspberrypi:~/grove.py/grove $ python grove_led.py
Hat Name = 'Grove Base Hat RPi'
Usage: grove_led.py <pin>
 <pin> could be one of below values
       in the pin column for GPIO function
   And connect the device to corresponding slot
==============
 pin | slot
==============
  5  | D5
 12  | PWM
 16  | D16
 18  | D18
 22  | D22
 24  | D24
 26  | D26
```

My environment is as follows:
- Raspberry Pi OS 11 (Bullseye)
- Python 3.9.2 (default version from Bullseye)
- pip 20.3.4 (default version from Bullseye)

I have setup I2c by:
sudo raspi-config > Interface Options > I2C

1, Can I know your environment as well? 
2, Could you try to reflash the SD card and start fresh?
3, Have you attached the Base HAT to the Raspberry Pi 4 in the correct orientation? 

Thank you.

chaeusler, 2022-03-28:
Hi @lakshanthad,

Thanks for your investigations!

1) 
```sh
pi@raspberrypi:~/grove.py/grove $ hostnamectl
   Static hostname: raspberrypi
         Icon name: computer
        Machine ID: ce513f1a44ff4c9dbd382eb66e08c51e
           Boot ID: b2e4734a4c6b474f95a6241b4ffb8db5
  Operating System: Raspbian GNU/Linux 11 (bullseye)
            Kernel: Linux 5.10.103-v7l+
      Architecture: arm
pi@raspberrypi:~/grove.py/grove $ python -V
Python 3.9.2
pi@raspberrypi:~/grove.py/grove $ pip -V
pip 20.3.4 from /usr/lib/python3/dist-packages/pip (python 3.9)
```

2)
Before reporting, I already started from scratch. Then I only executed the following commands:
```sh
git clone https://github.com/Seeed-Studio/grove.py
cd grove.py
sudo pip3 install .
```

3) 
yes

As I understand we have an identical setup.

lakshanthad, 2022-04-07:
Hello,

That is quite strange. Can I know how you enabled I2C?

Thank you.

priyankaprakasan, 2022-07-07:
Hi @chaeusler ,
Could you able to solve this problem?
I am also in the same situation.

duart38, 2022-09-21:
> Hi @chaeusler ,
> Could you able to solve this problem?
> I am also in the same situation.

Same issue here (I can see my i2c device using i2cdetect)

chaeusler, 2022-09-21:
The problem has been fixed with this commit: https://github.com/Seeed-Studio/grove.py/commit/a4a5c9bb00aa2bb5ee6d6d50098d6ddc05346140

duart38, 2022-09-21:
> My problem has been fixed with this commit: [a4a5c9b](https://github.com/Seeed-Studio/grove.py/commit/a4a5c9bb00aa2bb5ee6d6d50098d6ddc05346140)

Unfortunately this did not fix my issue. I'm still getting: ```
OSError: Please check if the I2C device insert in I2C of Base Hat
```

takanotume24, 2024-07-30:
I have created a script for installation and verified that it works with RasPi 4 (RasPi OS bullseye 32bit). 
Please refer to it if you like.
https://github.com/Seeed-Studio/grove.py/issues/62#issuecomment-2141651969

is-qian, 2024-09-21:
Refer to #73  .This issue is closed, If you have any questions, you can comment and reply.

closed 2024-09-21 by is-qian