Seeed-Studio/grove.py #53: Unable to install dependencies upm and mraa on Raspberry Pi Bullseye

Patrick257, 2022-02-06:
Possibly related to issue #46, I can't get the dependencies upm and mraa installed:

```
sudo apt install python3-mraa python3-upm
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-mraa : Depends: libjson-c3 (>= 0.10) but it is not installable
                Depends: libmraa1 but it is not going to be installed
                Depends: libpython3.5 (>= 3.5.0~b1) but it is not installable
 python3-upm : Depends: libmraa1 but it is not going to be installed
               Depends: libpython3.5 (>= 3.5.0~b1) but it is not installable
               Depends: libupm1 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

IcingTomato, 2022-02-07:
Because Python2 is obsolete and APT repository does not provide python-pip, so the Raspberry Pi OS releases which after `10 (Buster) 2020-08-20` cannot use `Online install` command. We also recommend using Python3 to all the users and developers.

You can use these commands down below.
```shell
sudo apt install git build-essential -y
git clone https://github.com/Seeed-Studio/grove.py
cd grove.py
sudo -H python3 -m pip install .
```
So that you don't need to install `mraa` and `upm` packages.

Patrick257, 2022-02-07:
Thanks!

However, if I don't install these packages, I'm not able to run the grove demos that require those packages. E.g.:

```
~/grove.py/grove $ python3 grove_3_axis_compass_bmm150.py
Traceback (most recent call last):
  File "/home/pi/grove.py/grove/grove_3_axis_compass_bmm150.py", line 19, in <module>
    from upm import pyupm_bmm150 as sensorObj
ModuleNotFoundError: No module named 'upm'
```

IcingTomato, 2022-02-13:
This problem is serious. We have been trying to deploy the environment. Please be patient。

Patrick257, 2022-02-13:
Thank you, @IcingTomato for the follow-up and looking forward to any further feedback on how to move ahead!

lakshanthad, 2022-03-08:
Hello,

I have tried to install mraa and upm by compiling from source. I was able to successfully install mraa, but not able to install upm, since there were some errors. Here is my attempt:

- Installing mraa

```
git clone https://github.com/eclipse/mraa
cd mraa
mkdir build && cd build
cmake ..
make 
sudo make install
```

- Installing upm
```
git clone https://github.com/eclipse/upm
cd upm
mkdir build && cd build
cmake ..
make 
sudo make install
```

I will look further into fixing upm install or finding a better installation method.

Thank you.

lakshanthad, 2022-03-20:
It seems that mraa and upm libraries and not maintained anymore. So you cannot use the mraa and upm related Grove modules. So I will close this issue for now. It can be opened later if a fix is found. 

Thank you.

closed 2022-03-20 by lakshanthad

iakovmarkov, 2023-09-13:
Why is this issue closed? The installation method is not working, the examples are not working either. 
What should the people who bought your stuff do?

patrickweyers, 2023-09-13:
Agreed. For what is' worth: My work around was a complete new system setup with Raspian Buster. This solves that particular issue for me but it creates other issues with component that would rely on an up-to-date OS.

Patrick257, 2023-09-13:
Agreed. For what is' worth: My work around was a complete new system setup with Raspian Buster. This solves that particular issue for me but it creates other issues with component that would rely on an up-to-date OS.

BasieDev, 2023-12-01:
DId anyone find a fix yet? Or is this a lost cause...

Macgyverops, 2023-12-22:
The problem is with newcomers don't see this on Amazon or Ebay or wherever they buy it.

Macgyverops, 2023-12-22:
I played around with this.  I found a workaround?  It's a freaking horrible one but I managed to do it.  

I tried `apt-get install python3-upm` but it was missing some kind of json manipulation library libjson-c which I downloaded directly and installed https://packages.debian.org/buster/armhf/libjson-c3/download

`wget http://http.us.debian.org/debian/pool/main/j/json-c/libjson-c3_0.12.1+ds-2+deb10u1_armhf.deb`
`sudo dpkg -i libjson-c3_0.12.1+ds-2+deb10u1_armhf.deb`

The crappy part is....I added buster back into the source list for apt.  So I don't know how long this fix will last once they deprecate buster.

`echo >> deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi`

After I did that retried upm
`apt-get install python3-upm`
Then mraa
 `apt-get install python3-mraa`

Macgyverops, 2023-12-22:
I tried this again from a fresh raspbian bullseye image (as of 12-05 image)

### Step 1
Run the installer for grove.py (the install.sh as it adds the seeeds source list)

### Step 1b
`echo "deb https://seeed-studio.github.io/pi_repo/ stretch main" | sudo tee -a /etc/apt/sources.list`

### Step 2
Add the buster repo into sources.list
`echo "deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi" | sudo tee -a /etc/apt/sources.list`

### Step 3
Install python3-upm
`sudo apt-get install python3-mraa python3-upm`

Macgyverops, 2023-12-22:
[image]

Woodz4242, 2025-02-15:
Just to add here - this is known issue now for so long. I can not even accept that this was "closed" in May 2022. How the &% you can close something like this - with no solution - when this is a commercial product sold? This is close to being fraud. Now 2025 and still no solution....

baorepo, 2025-02-17:
@Woodz4242  https://github.com/Seeed-Studio/grove.py/commit/7e5475a8511a9033e5d2f941bf1b13eafa5636ea We dealt with this issue in mid-2024, mraa and upm were not a good solution for us, and almost refactored the entire project to ensure software correctness while keeping the API unchanged.

iakovmarkov, 2025-02-17:
> [@Woodz4242](https://github.com/Woodz4242) [7e5475a](https://github.com/Seeed-Studio/grove.py/commit/7e5475a8511a9033e5d2f941bf1b13eafa5636ea) We dealt with this issue in mid-2024, mraa and upm were not a good solution for us, and almost refactored the entire project to ensure software correctness while keeping the API unchanged.

So what's the solution then?

baorepo, 2025-02-18:
@iakovmarkov  We use lower-level apis such as smbus and gpiod to replace the middle layers of mraa and upm.

Woodz4242, 2025-02-21:
Thanks for the comment. But where then they are the working installation instructions and examples for the Seeed Grove Base Hat.  And for Grove.py? 
I am still baffled that a product can be sold in a non working / non usable state...