Seeed-Studio/grove.py #56: Problem in i2c.py opening multiple i2c busses

maujac, 2022-04-27:
This issue was reported by @nikolozka in Issue #12.

The issue is still in the code and  I would like to submit a PR for it, however, I'm trying to understand the reasoning behind this.

Why is `instance` being used as a class attribute? It effectively makes the `Bus` class a singleton class.
Are there any advantages to doing this that I'm missing? Thanks.

lakshanthad, 2022-05-31:
Hello,

Could you please refer to the code here for i2c.py and try again?
https://github.com/Seeed-Studio/grove.py/commit/994c0467fe1e40b94e597536ab7760381b51cf2c#diff-52f36dd5ac4814c063e9adea411aa5764d3c963850457d2a022155f7172ada9c

Thank you

maujac, 2023-04-17:
I know it's been a long time but thanks for the suggestions. I tested the two suggestions and they both work in my case. Any ideas of when that code would be merged with `master`?

For those interested in getting this package working with multiple i2c buses immediately, as proposed by @nikolozka in #12 , I suggest the following changes in `grove/i2c.py`:

Modify lines 50 and 51:
```python
if not Bus.instance:
    Bus.instance = smbus.SMBus(bus)
```
To:

```python
if not self.instance:
    self.instance = smbus.SMBus(bus)
```
To find where `grove/i2c.py` is installed, try `pip show grove.py`.

maujac, 2023-04-17:
PR #66 created. Happy to discuss or modify it.

is-qian, 2024-10-16:
Thank you for your contribution, it has been merged into the latest code.

closed 2024-10-16 by is-qian