Seeed-Studio/grove.py #58: grove/grove_pwm_buzzer.py missing

trothe, 2022-06-10:
But both the https://github.com/Seeed-Studio/grove.py/blob/master/setup.py script as well as various documentation (e.g. https://wiki.seeedstudio.com/Grove_Base_Hat_for_Raspberry_Pi/#usage) prominently mention `grove_pwm_buzzer`. 

I cannot find `grove/grove_pwm_buzzer.py` it in the current master branch. Only a pull request from some time back. #55 
Am I doing something wrong or has the file been deleted for some reason?

Khaos66, 2022-11-03:
Found it online somewhere
```py
from __future__ import print_function

import time
from mraa import getGpioLookup
from upm import pyupm_buzzer as upmBuzzer

def main():

    print("Insert Grove-Buzzer to Grove-Base-Hat slot PWM[12 13 VCC GND]")
    
    # Grove Base Hat for Raspberry Pi
    #   PWM JST SLOT - PWM[12 13 VCC GND]
    pin = 12
    #
    # Create the buzzer object using RaspberryPi GPIO12
    mraa_pin = getGpioLookup("GPIO%d" % pin)
    buzzer = upmBuzzer.Buzzer(mraa_pin)
                                        
    chords = [upmBuzzer.BUZZER_DO, 
            upmBuzzer.BUZZER_RE, 
            upmBuzzer.BUZZER_MI,
            upmBuzzer.BUZZER_FA, 
            upmBuzzer.BUZZER_SOL, 
            upmBuzzer.BUZZER_LA,
            upmBuzzer.BUZZER_SI];
                                         
    # Print sensor name
    print(buzzer.name())
                                                     
    # Play sound (DO, RE, MI, etc.), pausing for 0.1 seconds between notes
    for chord_ind in range (0,7):
        # play each note for a half second
        print(buzzer.playSound(chords[chord_ind], 500000))
        time.sleep(0.1)
                                                                                          
    print("exiting application")
                                                                                               
    # Delete the buzzer object
    del buzzer
                                                                                                        
if __name__ == '__main__':
    main()                                                                                                     

```

is-qian, 2024-10-16:
The grove_pwm_buzzer has been re-added in the latest code. You can use it normally, and feel free to reopen this issue if you encounter any problems.

closed 2024-10-16 by is-qian