Seeed-Studio/grove.py #62: Can't Run Examples

dooley-ch, 2022-10-29:
It seems that mraa and upm libraries and not maintained anymore. So you cannot use mraa and upm related Grove modules. FIX THIS BLOODY ISSUE AND DO NOT CLOSE IT AGAIN UNTIL IT IS FIXED.

jbazala, 2022-11-02:
Hello, I have same issue, I bought Grove Base Kit for Raspberry Pi on August to learn how IoT works and then for using sensors and LCD in future project.. but nothings works.. all documentations, instructions and examples are out of date and I cannot figure how to get it working.. I am pretty disapointed. any sugestions? e.g. How did I get LCD running? I am using Raspberry Pi OS (bullseye).

jbazala, 2022-11-29:
I find a solution - I bought Arduino Uno and grove shield for it… And than I was able to run all grove sensors… But it is shame, that it does not work any more on RPi… Grove Base Kit for RPi is still on the stock, also at original e-shop, you can get it very easily. But it is very difficult to find out, that it will not work anymore. Especially for a newbie like me it is really time waste.

dooley-ch, 2022-11-29:
I’m glad you found a solution.

I’m actually retired and just play around with this stuff.  Perhaps when the mood takes me I will have a go at coding up a solution for the RPI.  I did it for some of the components for the BBC micro in the past.

https://github.com/dooley-ch/microbit-grove <https://github.com/dooley-ch/microbit-grove>

jonrot1906, 2023-03-31:
I actually found that to be very frustrating as well, as we have a few dozens of LCD displays and sensors sitting here, apparently without software support. After trying to build and install mraa and upm for hours without any luck [(script for mraa and upm installation)](https://gist.github.com/gaiar/3dc7a39261927d5e20e3), I decided to go on and found the [GrovePi repo from DexterIndustries](https://github.com/DexterInd/GrovePi). While not covering all sensors, I think that's a good starting point and helped me at least.

CurtisMJ, 2023-09-10:
Just in case someone finds this useful, I had managed to successfully compile UPM from source, but a few caveats needed resolving:

libmraa:
Seemed to go perfectly fine following the official build instructions on the Pi itself

UPM:
Needed to use clang to avoid errors with linking atomic symbols (GCC standard library for ARM is weird?? Really don't know)
Needed to manually copy the python parts like "pyupm.py" to the dist-packages/upm folder,__init__.py seemed blank for some reason 
Needed to export LD_LIBRARY_PATH to the upm lib folder before running python examples. Could probably be solved by setting the install prefix correctly for Debian when running cmake
All of this was done on Raspian (RPI OS?) 11 bullseye, with Python 3.9. I can now use the Grove LCD module (the white on blue one) just fine with the Grove hat. This was on a Raspberry Pi Zero, so I left it to compile overnight, cross-compiling may have been easier... but it worked lol

takanotume24, 2024-02-25:
> Just in case someone finds this useful, I had managed to successfully compile UPM from source, but a few caveats needed resolving:
> 
> libmraa: Seemed to go perfectly fine following the official build instructions on the Pi itself
> 
> UPM: Needed to use clang to avoid errors with linking atomic symbols (GCC standard library for ARM is weird?? Really don't know) Needed to manually copy the python parts like "pyupm.py" to the dist-packages/upm folder,__init__.py seemed blank for some reason Needed to export LD_LIBRARY_PATH to the upm lib folder before running python examples. Could probably be solved by setting the install prefix correctly for Debian when running cmake Needed to pass "DWERROR=OFF" because clang is more pedantic than whatever they usually compile with Also passed "DSWIGNODE=OFF" because I wasn't using Node All of this was done on Raspian (RPI OS?) 11 bullseye, with Python 3.9. I can now use the Grove LCD module (the white on blue one) just fine with the Grove hat. This was on a Raspberry Pi Zero, so I left it to compile overnight, cross-compiling may have been easier... but it worked lol

@CurtisMJ  Thanks so much for sharing useful information. I too was able to get it to work with RasPi 4 (RasPi OS bullseye 32bit) + Grove Base Hat + Grove tempareture sensor, but not with RasPi OS bookworm, probably due to its dependency on Python2. It is not working on RasPi OS bookwarm, probably because it depends on Python2.

takanotume24, 2024-05-31:
I have published a bash script to solve this dependency issue. Please feel free to use it as a reference if you find it helpful. https://github.com/takanotume24/grove_py_installer

is-qian, 2024-09-22:
Refer to https://github.com/Seeed-Studio/grove.py/issues/73 .This issue is closed, If you have any questions, you can comment and reply.

closed 2024-09-22 by is-qian