Seeed-Studio/grove.py #65: Grove Kit does not install on Raspberry Pi 400 - 8gb

sealman38, 2023-02-21:
curl -sL https://github.com/Seeed-Studio/grove.py/raw/master/install.sh | sudo bash -s -
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:3 https://seeed-studio.github.io/pi_repo stretch InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
python3-pip

E: Package 'python-pip' has no installation candidate
dpkg-query: package 'python-pip' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
python3-pip

E: Package 'python-pip' has no installation candidate
dpkg-query: package 'python-pip' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
python3-pip

E: Package 'python-pip' has no installation candidate
dpkg-query: package 'python-pip' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-pip is already the newest version (20.3.4-4+rpt1+deb11u1).
The following package was automatically installed and is no longer required:
libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
I2C interface enabled...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-rpi.gpio is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-rpi.gpio' has no installation candidate
dpkg-query: package 'python-rpi.gpio' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-rpi.gpio is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-rpi.gpio' has no installation candidate
dpkg-query: package 'python-rpi.gpio' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-rpi.gpio is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

NicolasAppsDevelopment, 2024-01-20:
"Because Python2 is obsolete and APT repository does not provide python-pip, so the Raspberry Pi OS releases which after 10 (Buster) 2020-08-20 cannot use Online install (`curl -sL https://github.com/Seeed-Studio/grove.py/raw/master/install.sh | sudo bash -s -`) command. We also recommend using Python3 to all the users and developers."

Use the Step by step installation instead:
```
git clone https://github.com/Seeed-Studio/grove.py
cd grove.py
# Python2
sudo pip install .
# Python3
sudo pip3 install .
```

caution: If you are using Raspberry Pi with Raspberrypi OS >= Bullseye, you have to use this command line only with Python3.

is-qian, 2024-09-20:
It seems to be an issue related to the use of python-pip, which currently does not occur in the latest v0.8 version. You can try again.

closed 2024-09-20 by is-qian