Seeed-Studio/grove.py #68: Raspberry Pi 4 and hat not working

achapkowski, 2023-06-28:
I have two different raspberry Pi 4 b and each has the same issue.

'No modules name upm'

I followed the instructions on #46 but it didn't work.

I also followed all the instructions on the wiki as well.

I have I2C enabled.  
Python 3.9.2
OS: Linux Debian GNU Linux 11 

It was the 64-bit image provided by Raspberry Pi folks.

Please advise.

Thank you

alansill, 2024-01-09:
I followed the instructions in the last comment in the closed issue https://github.com/Seeed-Studio/grove.py/issues/53 and was able to get pas the installation failures. 

Note this is *NOT* an ideal solution!

alansill, 2024-01-09:
Note I was using the 32-bit Raspian image.

takanotume24, 2024-07-30:
I have created a script for installation and verified that it works with RasPi 4 (RasPi OS bullseye 32bit) + Grove Base Hat + Grove tempareture sensor. 
Please refer to it if you like.
https://github.com/Seeed-Studio/grove.py/issues/62#issuecomment-2141651969

is-qian, 2024-08-20:
#73

is-qian, 2024-08-20:
Refer to #73 .This issue is closed, If you have any questions, you can comment and reply.

closed 2024-08-20 by is-qian