Seeed-Studio/grove.py #73: Support for Raspberry Pi OS 12 (bookworm)

tuxgasy, 2024-05-03:
Hello,

The install script don't work with Bookworm. Is there any solution ?

Thanks.

Henry-baobao, 2024-05-15:
I have the same issue, too. No one maintain this reposity anymore? How can i solve this problem in latest Raspberry Pi OS?

Henry-baobao, 2024-05-15:
@tuxgasy Hi, did you fix the problem?

tuxgasy, 2024-05-16:
No. Finally, I use [gpiozero](https://gpiozero.readthedocs.io/en/latest/) directly. For the basic module (like the LED and the button), it works.

s-light, 2024-07-09:
after a look through the install.sh i think to fix this is a bigger rewrite / write a new install script.
and after searching some more in the issues i found:
- #70
- #71 
- #46 
- #68 
- #53 

has anybody tried the installation instructions from the wiki in a current fresh rpi os image?
https://wiki.seeedstudio.com/Grove_Base_Hat_for_Raspberry_Pi/#installation

> git clone https://github.com/Seeed-Studio/grove.py
> cd grove.py
> sudo pip3 install . --break-system-packages

dsrc12, 2024-07-23:
> Hello,
> 
> The install script don't work with Bookworm. Is there any solution ?
> 
> Thanks.
There is another issue regarding the Grove HAT compatibility with Bookworm.
 
Seeed updated the wiki for Raspberry Pi Model 5 using Bookworm on Fri 19 Jul24 but there is still an issue. The Grove HAT has two versions: One (STM32) uses i2c address x04 and the latter one (MM32) uses an i2c address x08. Bookworm follows the i2c standard and does not recognise i2c addresses less than x08. So you can get the Grove HAT to work in Bookworm has long as it is based on the MM12 microcontroller with an i2c address of x08.
 The basic problem is that Seeed do not support software after the launch date of a product - they are a 'just in time' company and focus only on new products. The Grove HAT spans many generations of OS and is the Grove python library is incompatible with the latest version of the OS (bookworm).
 
Another issue is  Seeed Studio's poor use of "break-system packages". This is very crude and should not be used under Bookworm. It would have been better if Seeed Studio used a virtual env for installing the Grove python framework but the reference.

quietCoder500, 2024-08-01:
Anyone with this issue, could you try out my Pull request #76. It removed support for three grove modules but the code should run. If the install doesn't work, can you post the error there or here so I can look at it. I tried the install on a Pi 4 and it succeed (not without warnings thought), probably thanks to PR #74.

is-qian, 2024-08-20:
In this pull request #77 , the UPM and MARR libraries have been replaced, and the relevant Grove tests have been passed in the Bookworm.

closed 2024-08-20 by is-qian

Woodz4242, 2025-02-15:
I find it quite "indecent" to (still) sell a product - marketing it as "Kit for Raspberry Pi"", knowing that it can not run on the latest Pi/operating systems. Sure they could claim you can easily go back to older OS - but this is not an option due to several things, like performance on newer systems, python3 updates, security, etc., etc. This is just bad "habit" or blunt money making. I am really disappointed here with Seeed.
I have fought now hours and hours - and can only get the simplest things to work, like the temp/humi sensor. But not even the LED from the Grove Base Kit works properly on a Raspi5 with Bookworm.
Michael