Seeed-Studio/grove.py #78: rove-Laser_PM2.5_Sensor-HM3301 sporadic readings

tim-oe, 2024-10-21:
i adapted the file
https://github.com/Seeed-Studio/grove.py/blob/master/grove/grove_PM2_5_HM3301.py
and did a slight adaptation to breakout the data from the reading
https://github.com/tim-oe/WeatherWatch/tree/main/weatherwatch/sensor/aqi

the sensor is being read every 15 minutes and they seem to jump up and down.
[aqi_sensor.csv](https://github.com/user-attachments/files/17452948/aqi_sensor.csv)

Lesords, 2024-11-08:
> i adapted the file https://github.com/Seeed-Studio/grove.py/blob/master/grove/grove_PM2_5_HM3301.py and did a slight adaptation to breakout the data from the reading [...]

@is-qian Help with this one.

tim-oe, 2024-11-10:
i have it mounted in a vertical position, wondering if that would impact it?
https://github.com/tim-oe/WeatherWatch/blob/main/docs/img/aqi.jpg

is-qian, 2024-11-18:
I got a Grove, and indeed noticed abnormal fluctuations in readings, which seemed to have nothing to do with installation. I will further investigate it. Thank you for your question.

tim-oe, 2024-11-22:
thanks for looking into this, iv'e used 2 different sensors with same behavior
i get a lot of crc failures before i can get a sensor reading.
here's a graph of a weeks worth of readings, this is discarding readings over 1000k
[image]

is-qian, 2024-11-25:
How long is your delay [here](https://github.com/tim-oe/WeatherWatch/blob/main/weatherwatch/sensor/aqi/Hm3301Reader.py#L26)? After I changed the delay of [here ](https://github.com/Seeed-Studio/grove.py/blob/master/grove/grove_PM2_5_HM3301.py#L120)to 1s, the crc error didn't appear again when it was powered on for the first time. Strangely, I didn't encounter reading jitter after that.

tim-oe, 2024-12-02:
i've tried delays from 1 to 5 seconds, in between the write and read commands.
https://github.com/tim-oe/WeatherWatch/blob/main/weatherwatch/sensor/aqi/Hm3301Reader.py

what is not clear to me is that in doing this periodically i issue both the write and read when reading data.
essentially doing what's being done in the sample code in an infinite loop every 10 minutes

looking at the data sheet though the write and read address don't line up:
(see Appendix 2: IIC Communication Protocol)
https://files.seeedstudio.com/wiki/Grove-Laser_PM2.5_Sensor-HM3301/res/HM-3300%263600_V2.1.pdf

the sample and my code both use 88, the appendix shows write at 80 and read at 81

is-qian, 2024-12-16:
The i2c address here is 7-bit (0x40), with the last bit being the R/W bit. When using write in smbus2, the R/W bit is set to 0, so adding the 7-bit address results in 0x80 as specified in the datasheet. When using read, the resulting address is 0x81.

closed 2025-01-30 by github-actions[bot]